DroidPlanner/Tower #1351: make Speech sentence editable

Pomaroli, 2015-01-03:
Could it be possible to make the speech sentence editable ?

ne0fhyk, 2015-01-03:
@Pomaroli you can suggest new or improvements to the current speech sentences. Making them editable seems like too much work for a feature that would be rarely used.

Pomaroli, 2015-01-03:
Hi,
yes i understand, is there no way to edit them on XMl file ?
The problem are the trasnlations in other languages, an example, for German it should not say stabilize but something like "Manuel" otherwise the speech is laughable. I can do the senteces for german if you can give me the english default.

closed 2015-01-06 by phmagic

ne0fhyk, 2015-01-06:
@Pomaroli The devs contributing to the german translation are @zumbrunnen and @squilter . Perhaps you might want to check with them regarding the correct term.

zumbrunnen, 2015-01-06:
Hi @Pomaroli 

A lot of things (e.g. the flight modes, RC/tuning params) are still hard-coded in English. In order to translate them, we first have to change the code to use localized variables. After that, it is as simple as populating the XML files with the localized contents.

If you point me to the right direction, I'll change the code and translate stuff to German. Please give me your priorities. :wink: Shall I cast an eye on the spoken flight modes to begin?

Pomaroli, 2015-01-19:
Hi,
yes this is a good idea. Sorry for delay.