DroidPlanner/Tower #1354: Droneshare screen while I'm flying

andrea25, 2015-01-06:
Found I bug. I think. Probably. I was flying in a park near my office. I was testing droid planner 3. I decided to go in the air by clicking Take-Off on the the tablet. And it is all right, I look at the tablet to click land but a screen about drone share appeared on my screen. So I can not click land. And I think ” what if I'm in a hurry and I need to land immediately? Isn't this dangerous?".
And another time, I was using follow me, I was running and the drone was chasing me. I see that it's better to land the drone or to stop the follow me because I was approaching some trees. I look at my tablet and I see again the drone share screen. I need to land so I used the normal radio because with the drone share screen on the tablet it was difficult. An I think again "Isn't this dangerous?"

My opinion is: while copter is in the air the drone share screen should not appear in any case.
[image]

ne0fhyk, 2015-01-06:
@andrea25 I replied to your post on the g+ community. Let's continue the conversation there to figure out how to make it less of an annoyance.

closed 2015-01-06 by ne0fhyk

andrea25, 2015-01-06:
OK, thank you! I will reply ASAP on G+.