DroidPlanner/Tower #1358: Parameter editing appears broken on Android L

JamesHarrison, 2015-01-11:
Hi,

On Android L, using the default Google keyboard, if I put my cursor in any parameter field, the cursor is forced to be in front of the value, making it impossible to delete. I can't edit the parameter in any way, or move the cursor. Old Droidplanner does not exhibit this bug, and works as expected, so I'm currently hopping between clients to edit parameters when needed in the field.

ne0fhyk, 2015-01-11:
@JamesHarrison that issue has been fixed in the latest beta of the app. You can find more info about the latest beta in the google+ forums (https://plus.google.com/communities/109498440846585781402/stream/ae684de5-25c7-49cd-9c47-0d9d37481737), or in the mailing list (https://groups.google.com/forum/#!topic/droidplanner/2Z37sE1tjS4).

closed 2015-01-11 by ne0fhyk

MJBeaubien, 2015-01-11:
Has the beta version been updated with the finalized edits on the Google Play Store?

ne0fhyk, 2015-01-11:
@MJBeaubien which edits are you referring to? 

If you're referring to the fix for parameter editing, yes, the beta for DP3 contains the fix.

JamesHarrison, 2015-01-11:
The beta signup form linked to on the Github repo for beta 3 doesn't appear to be accepting signups.

Good to know this is fixed and will presumably be hitting the Play store soon, in any case!

ne0fhyk, 2015-01-11:
@JamesHarrison due to a play store requirement, you need to be either part of the droidplanner mailing list, or of the droidplanner google+ community to get access to the beta.