DroidPlanner/Tower #1364: Please add custom maps / make it easy to add maps.

AndKe, 2015-01-19:
Please observe how easy OruxMaps have made it to add custom maps, a small selection is here, the last one, Norwegian topographic maps are very important to us in this country - it show proper altitudes.

I tried to look into the source to add it today, forund the googlemaps folder, but it was way to hard to understand how to properly add a map.

   <onlinemapsource uid="0">
        <name>Google Maps</name>
        <url><![CDATA[http://mt{$s}.google.com/vt/lyrs=m@121&hl={$l}&x={$x}&y={$y}&z={$z}]]></url>
        <minzoom>0</minzoom>
        <maxzoom>19</maxzoom>
        <projection>MERCATORESFERICA</projection>
        <servers>0,1,2,3</servers>
        <httpparam name=""></httpparam>
        <cacheable>1</cacheable>
        <downloadable>1</downloadable>
        <maxtilesday>0</maxtilesday>
        <maxthreads>0</maxthreads>
        <xop></xop>
        <yop></yop>
        <zop></zop>
        <qop></qop>
        <sop></sop>
    </onlinemapsource>
 
    <onlinemapsource uid="2">
        <name>Google Terrain</name>
        <url><![CDATA[http://mt{$s}.google.com/vt/lyrs=t,r&hl={$l}&x={$x}&y={$y}&z={$z}]]></url>
        <minzoom>0</minzoom>
        <maxzoom>15</maxzoom>
        <projection>MERCATORESFERICA</projection>
        <servers>0,1,2,3</servers>
        <httpparam name=""></httpparam>
        <cacheable>1</cacheable>
        <downloadable>1</downloadable>
        <maxtilesday>0</maxtilesday>
        <maxthreads>0</maxthreads>
        <xop></xop>
        <yop></yop>
        <zop></zop>
        <qop></qop>
        <sop></sop>
    </onlinemapsource>
    <onlinemapsource uid="3">
        <name>OpenStreetMap Cyclemap</name>
        <url><![CDATA[http://{$s}.tile.opencyclemap.org/cycle/{$z}/{$x}/{$y}.png]]></url>
        <website><![CDATA[<a href="http://www.opencyclemap.org/">OpenStreetMap Cyclemap</a>]]></website>
        <minzoom>0</minzoom>
        <maxzoom>17</maxzoom>
        <projection>MERCATORESFERICA</projection>
        <servers>a,b,c</servers>
        <httpparam name="User-Agent">{om}</httpparam>
        <cacheable>1</cacheable>
        <downloadable>1</downloadable>
        <maxtilesday>0</maxtilesday>
        <maxthreads>0</maxthreads>
        <xop></xop>
        <yop></yop>
        <zop></zop>
        <qop></qop>
        <sop></sop>
    </onlinemapsource>
     
        <onlinemapsource uid="4">
        <name>OpenStreetMap Mapnik</name>
        <url><![CDATA[http://{$s}.tile.openstreetmap.org/{$z}/{$x}/{$y}.png]]></url>
        <website><![CDATA[<a href="http://www.openstreetmap.org">OpenStreetMap Mapnik</a>]]></website>
        <minzoom>0</minzoom>
        <maxzoom>18</maxzoom>
        <projection>MERCATORESFERICA</projection>
        <servers>a,b,c</servers>
        <httpparam name="User-Agent">{om}</httpparam>
        <cacheable>1</cacheable>
        <downloadable>1</downloadable>
        <maxtilesday>0</maxtilesday>
        <maxthreads>0</maxthreads>
        <xop></xop>
        <yop></yop>
        <zop></zop>
        <qop></qop>
        <sop></sop>
    </onlinemapsource>
     

    <onlinemapsource uid="13">
        <name>Statkart Topo2 (NO)</name>
        <url><![CDATA[http://opencache.statkart.no/gatekeeper/gk/gk.open_gmaps?layers=topo2&zoom={$z}&x={$x}&y={$y}]]></url>
        <website><![CDATA[<a href="http://statkart.no">statkart.no</a>]]></website>
        <minzoom>0</minzoom>
        <maxzoom>17</maxzoom>
        <projection>MERCATORESFERICA</projection>
        <servers>0,1,2,3</servers>
        <httpparam name="User-Agent">{om}</httpparam>
        <cacheable>1</cacheable>
        <downloadable>1</downloadable>
        <maxtilesday>0</maxtilesday>
        <maxthreads>0</maxthreads>
        <xop></xop>
        <yop></yop>
        <zop></zop>
        <qop></qop>
    </onlinemapsource>

bys1123, 2015-01-22:
Yes, it's hard to use google maps in our country.

ne0fhyk, 2015-01-22:
The code is structured to allow easy integration of additional map providers. I don't have the bandwidth to add other map providers at this point, but I will gladly accept PRs for that.

AndKe, 2015-01-22:
I added Norwegian maps to Oruxmaps, I also did it to APm Planner 2.0  - that was easy.
i am completely lost at the map-related files of droidplanner, huge files, lots of code, and hard to find/understand the logic how/where the maps are fetched.

maybe a short tutorial (if something that complex can be covered in that) - would allow us to add good maps ?

bys1123, 2015-01-28:
I notice that the code is too simple to support lot of map providers that not use this format.

AndKe, 2015-01-29:
@bys1123 - I don't know that - but all kinds of google maps are accessible that way, Bing, Open* , even satellite weather data and Yandex, so I've never missed a site.
It's simply tile-reading , which is what most seem to use.

bys1123, 2015-01-29:
[image]
Bing seems each tile have a perticuler number, not z x y . In MP, Bing maps providers of GMap.NET is very complicated.

AndKe, 2015-01-29:
true, I stand corrected, Bing does not work for me anymore, it did before, with the code;
'''   <onlinemapsource uid="1009">
                <name>Bing Hybrid</name>
                <url><![CDATA[http://ecn.t0.tiles.virtualearth.net/tiles/h{$q}.jpeg?g=604&mkt=en-us&n=z]]></url>
                <minzoom>1</minzoom>
                <maxzoom>19</maxzoom>
                <projection>MERCATORESFERICA</projection>
                <website><![CDATA[© 2011 Microsoft | <a href="http://go.microsoft.com/fwlink/?LinkId=74170">Privacy</a> | <a href="http://g.msn.com/0TO_/enww">Legal</a>]]></website>
                <httpparam name=""></httpparam>
                <cacheable>1</cacheable>
                <downloadable>1</downloadable>
                <maxtilesday>0</maxtilesday>
                <maxthreads>0</maxthreads>
                <xop></xop>
                <yop></yop>
                <zop></zop>
                <qop></qop>
                <sop></sop>
        </onlinemapsource>
'''
Anyway, I don't miss it, nor would I encourage anyone to get near the .NET lock-in-mess.

ne0fhyk, 2015-02-05:
@bys1123 from the conversation, it seems you are planning to write the code for an additional `DPMapProvider` to support the feature mentioned by `AndKe`.
If i'm correct, feel free to let me know if you need any pointers.

bys1123, 2015-02-06:
Sorry, maybe I'm not say that clearly. I'm trying use DPMapProvider add a map. I think this is enough. 
Writing code for the feature in this issue is too difficult for me.
Thank you.

SkiTiSu, 2015-02-10:
Hi, @ne0fhyk ,I'm in @bys1123 's team. 
Chinese user's experience is very bad because we can’t use Google without proxy. We used to add AMap and offset correction for it and GoogleChinaMap in MP. Now we are trying to add Baidu Map SDK in DPMapProvider. Baidu is a sponsor for Dronecode Fondation.
Now here’s a problem, the Baidu Map API needs the SHA1 of Signing Certificate of Tower.
Will we be able to get the SHA1 of Signing Certificate? Thanks a lot.

ne0fhyk, 2015-02-10:
@SkiTiSu could you provide a pull request with the Baidu map provider implemented, as well as instructions on how to create an account for the baidu map sdk.

bys1123, 2015-02-10:
@ne0fhyk OK, we thought its all Chinese pages maybe too difficult to do it by yourself before. Your suggestion sounds more reasonable. we'll try to make some picture instructions and pull request in 36 hours.

ne0fhyk, 2015-02-10:
@bys1123 sounds great, thanks!

ne0fhyk, 2015-09-26:
Support for custom tile providers was added in version 3.2.0.

closed 2015-09-26 by ne0fhyk