DroidPlanner/Tower #1377: Wrong camera angle for Gopro Hero3+ in function Survey

lorbass, 2015-02-02:
The applicated camera angle in Droidplanner is 10 degrees (calculated without overlaping)
But the angle of the Gopro in wide mode ist 150 degrees.
Is it perhaps a decimal error and should be 100 degrees?
And there is no more the possibility to change angle and overlaping in Version 2.8.6 RC3

azrin1972, 2015-02-02:
Since go pro hero 3 support 3 different angle, maybe we can accommodate the different angle or just ask user to change to narrow angle (no fish eye) before flying.

lorbass, 2015-02-02:
Narrow mode of Gopro has an angel of 90 degrees. But Droidplanner is 
drawing 10 degrees.
Still far away from Gopro. Of course narrow angle (90 degrees) is better 
as you wrote because of less fish eye effect.
As example: At 50m atltitude Droidplanner is planning 11 times to cross 
the length of a football field which has 100m.