DroidPlanner/Tower #1378: Cannot finish the compass calibration

andrea25, 2015-02-03:
I had some issues with compass calibration in droid planner 3 beta.
I can start it, it seems to work, the point appear on the screen, but I can not finish it with success.
I can cancel it, but I can not do it in the right way.

With mission planner I have no problem with compass calibration.

Does anyone have the same problem with droid planner 3 beta?

zumbrunnen, 2015-02-12:
I'm having a related issue: I cannot access/start compass calibration at all. Only "Accelerometer Calibration" is available. 

@ne0fhyk is this feature gone temporarily? 

@andrea25 Ensure you are doing the right movements during calibration. Check Chris Anderson's "Calibration Dance" on Youtube: https://www.youtube.com/watch?v=-_mjfPlHL9o 

Doing his funky dance, I've always managed to succeed calibrating. Unfortunately, I'm not in the groove anymore as there is no more "Compass Calibration" in Tower anymore (at least for me).

ne0fhyk, 2015-02-12:
@zumbrunnen it's been removed from the production release until it's fully working. It's still available in the beta releases (although they're no current beta out at the moment.. still working on the next one).

zumbrunnen, 2015-02-12:
@ne0fhyk Thank you for the quick reply :+1:

andrea25, 2015-02-13:
@zumbrunnen Thank you for reply =) The "dance" appear to be correct. In matter of fact the same "dance" works very well with mission planner, but it do not works with droid planner 3/tower beta.

By the way I see that the compass feature was removed, so: no more problem =P

closed 2015-05-26 by ne0fhyk