DroidPlanner/Tower #1384: wrong focal length for multiple cameras

martinjuju, 2015-02-05:
Hi
doing a survey with a sony nex 5 (25mm) should be roughly the same as with a sony rx100M3 (24mm).
The number of pictures and survey size should theoretically be slighly lower for the rx100 given the wider focal. However, a short survey with the sony rx100M3 provides 42 pictures and 6 strips whereas the nex 5 (25mm) gives 19 pictures and 4 strips.

After a bit of hunting on the source of the problem, it iseems that the .xml files in the CameraInfo folder are wrong for the rx100M3.
Focal length listed in the .xml files for rx100M3 is the 35mm equivalent and not the true lens length.
It should be 9.0 and 26.4 in the 24 and 70mm files respectively.

If you want I can post updated files and check info the .xml of other cameras.
Could I also post files for new cameras or other setup (e.g. gopro medium angle)?
Keep up the great work

ne0fhyk, 2015-02-05:
@martinjuju thanks for the bug report! Yes, updated files, as well as files for new cameras would be great!

ne0fhyk, 2015-02-24:
Fixed by PR #1412

closed 2015-02-24 by ne0fhyk