DroidPlanner/Tower #1385: Clean up permissions

squilter, 2015-02-06:
Since the back-end is now handled by 3dr-services, I don't believe that Tower needs bluetooth, etc. anymore.

ne0fhyk, 2015-02-06:
Could you list the permissions you think are not needed anymore. The bluetooth ones are still needed because Tower needs to be able to present the user with a list of available bluetooth devices.
Once a bluetooth device is selected, it passes it bt address to 3DR Services.

squilter, 2015-02-06:
Scanning needs `BLUETOOTH_ADMIN`, but `BLUETOOTH` is for connecting.  Or does scanning need both of those?

Where are `ACCESS_NETWORK_STATE` and `ACCESS_WIFI_STATE` used?

ne0fhyk, 2015-02-06:
`BLUETOOTH` is required to list paired bluetooth devices, while `BLUETOOTH_ADMIN` is required to discover and pair with new devices.
I'm investigating if we're still using `ACCESS_NETWORK_STATE` and `ACCESS_WIFI_STATE`.

ne0fhyk, 2015-02-06:
`ACCESS_NETWORK_STATE` is used by the google maps api.
And `ACCESS_WIFI_STATE` will be used in the future to help with offline maps caching.

closed 2015-02-06 by ne0fhyk