DroidPlanner/Tower #1389: Low batt failsafe Bug

Cyberpilot360, 2015-02-07:
On droidplanner 3 when battery goes below the set failsafe (3.5v per cell in this case) and the copter is in flight the options of RTL, PAUSE and FOLLOW disapear and i'm only able to choose takeoff and disarm- which has been tested and actually disarms the copter in the air!! Which is very dangerous. 
I have noticed that on both DP2 and DP3 beta, I have no low battery failsafe action set (no land or RTH) and I'm using an S3mini and LG V500 tab.

ne0fhyk, 2015-02-09:
@Cyberpilot360 Could you give more info about the vehicle you were flying with, such as what type of hardware, and autopilot firmware.
Also could you include the parameters list for that vehicle.

thanks.

ne0fhyk, 2015-02-12:
@Cyberpilot360 the issue was in [3DR Services v1.2.6.beta](https://github.com/DroidPlanner/3DRServices/pull/51).

closed 2015-02-12 by ne0fhyk

Cyberpilot360, 2015-02-12:
It was a pixhwawk with 3.2.1 QUAD X FW. Sorry I missed your reply. 
I can't send you the param list right now, do you still need them? 
Where can I get acces to the fixed beta version? 
Thanks!

ne0fhyk, 2015-02-12:
@Cyberpilot360 I was able to reproduce the behavior, so the param list won't be needed.
To get access to the 3DR Services beta:
 1. You'll need to join the 3DR Services [google+ community](https://plus.google.com/communities/109484130178804610965) or [mailing list](https://groups.google.com/forum/#!forum/3dr-services).
 2. Follow the instructions at the following link: https://play.google.com/apps/testing/org.droidplanner.services.android

**Note**: If you do, bear in mind you'll be using a **beta** version, and as such bugs are expected.