DroidPlanner/Tower #1391: Feature Request - Live HD video playback

PaulHam211, 2015-02-09:
Hi

Would it be possible for a toggle between map view and live HD video playback from a RTMP:// stream?

I am working on low latency HD video downlink with the raspberry pi, currently MXPlayer can play the stream but it would be good to implement into Droidplanner/Tower.

andersonpaac, 2015-02-09:
Impossible to implement with the minimal connection speed on telemetry radios

squilter, 2015-02-09:
I assume the OP was suggesting that telemetry remains over the radios, but that video goes over wifi.  I think the easiest way to accomplish this would be to install something like VLC for android to play the video stream, and have tower open next to it.  I don't think tower has multi-window support, but it shouldn't be as hard to add.

ne0fhyk, 2015-02-09:
It's possible to have the video playback native to Tower. I implemented a video decoding library for h264 over udp, so it could be repurposed for that functionality. The caveat is that the library only supports devices running on Android 4.3 and up.

@PaulHam211 are you familiar with java programming? Video decoding integration is usually no easy task, and I currently wouldn't have the bandwidth to handle it all myself.

Linjieqiang, 2015-02-10:
sounds great！

PaulHam211, 2015-02-11:
Yes @squilter is correct, Video downlink through WIFI and telemetry through telemetry radios (although the WIFI can support the telemetry link also)

@ne0fhyk Unfortunately my knowledge of java programming is limited/non-existent. I am managing to run bash scrips on the Raspberry Pi but this is it.

It would be great to have this implemented into tower if somebody gets the time to work it out. I know DJI have a expensive lightbridge system which is basically the same idea.

JonoParrott, 2015-03-07:
Just wanted to reopen this long requested feature as VR Pad Station has already accomplished this low latency video streaming feature in their open-source code which is here https://code.google.com/p/vr-pad-station/source/checkout

I believe it would be an instance of someone porting the VLC library encode/decode into Tower. I'd be happy to do it....but I haven't a clue about app development :(

mhkabir, 2015-03-07:
Hi all,
All we need is a Gstreamer interface (already implemented in VR Pad Station) which can display a stream. The video interface into the tablet / phone would be WiFi, which could be run directly from the vehicle, or bridged from a long range link using a router.
VirtualRobotix already have this in their GCS, so it'd be easy to port over.
@arthurbenemann

rubenverhoef, 2015-04-24:
a live video from gstreamer would be great!

ne0fhyk, 2015-09-26:
Version 3.2.0 supports live hd video playback from a Solo vehicle.
Support for other vehicles is considered for the following releases.

closed 2015-09-26 by ne0fhyk