DroidPlanner/Tower #1394: Droidplanner 3 (Tower) RC9 Pixhawk output changes with flight mode

Bradatech, 2015-02-11:
I have some interesting behavior with Tower.  I've got a gimbal connected to RC9 on a Pixhawk which has the tilt configured on channel 6 as described on the wiki for Follow Me and ROI to work.  Works great on DP2 but DP3, any changes to the flight mode will cause the gimbal to point straight ahead (horizon) and then lock in place until I switch modes again.  Then I can change the tilt manually just fine.  This is in DISARMED mode.  Haven't flown it yet on DP3.

Running AC3.2 latest build.

[image]
[image]

rmackay9, 2015-02-24:
Could you check what your MNT_MODE is when you connect?
It should be "3" normally which means under RC control.  Maybe it's somehow been set to "4" or some other number.

ne0fhyk, 2015-02-24:
@Bradatech as mentioned in #1406, the issue you encountered was fixed in  [`3DR Services version 1.2.9`](https://github.com/DroidPlanner/3DRServices/pull/63).

closed 2015-02-24 by ne0fhyk

Bradatech, 2015-02-24:
@rmackay9 Randy, the value is 3.

@ne0fhyk Fredia, nice work!  all is well with the gimbal now.  The 3DR services update came through this morning via GooglePlay. Also, the Android Wear portion is now connecting and works very well.

Great job to you guys and the dev group.  The Pixhawk's future looks very promising and can't wait for optical flow :)