DroidPlanner/Tower #1396: OnePlus One Bluetooth issue

lvale, 2015-02-11:
Hi.

Current setup is on PixHawk hexa fw 3.2.1 using the 3DR Bluetooth dongle connected to telem2 port.

Mobile phone is a OnePlus One running stock fw (Cyanogen Mod 44s-Android 4.4.4)

Although Andropilot runs perfectly and connects easily, Tower 3.1.2 won't connect via Bluetooth.

Error message:

Connection failed: read failed, socket might closed or timeout, read ret: -1

3DR Services is version v1.2.5 beta.

Something else 3DR Services is annoying as hell because it won't stop and disappear from my notification bar, so I usually force it to quit.

ne0fhyk, 2015-02-12:
@lvale have you tried to forget the default bluetooth device (in the settings screen), and reselect it in the selection list. Also depending on the bluetooth dongle, it might need to be paired first with the phone.

The 3DR Services notification bar only appears when an app is connected to its background service. It disappears as soon as they're no apps left connected to it.
You can test that by opening Tower (upon which the status notification will appear), and closing it using the back button (which should remove the status notification).

lvale, 2015-02-12:
Issue two: 3DR Services really closes but takes about one minute and half to disappear :) So I was too fast :)

Issue one: Using the 3DR Bluetooth dongle, so when connecting to the phone it requires that pairing is done with pin. Checked that.

Sequence of procedures:

1. Remove HC-06 device from Paired devices list on the phone.
2. Only other devices that are paired with the phone, but not present here, are my car's Bluetooth connection and a OBD2 dongle also from my car. i.e. they appear on the paired devices list but not active.
3. Power up PixHawk.
4. On the phone Bluetooth search for devices. It finds the HC-06 and when selected asks for the pin to pair.
5. Enter pin, and the HC-06 appears as paired.
6. Start Tower, check the settings Telemetry connection type is set to Bluetooth.
7. On Tower main screen press the Connect button on the bottom. Phone says "connecting" and after about 4/5 seconds the error message is displayed briefly:
Connection failed: read failed, socket might closed or timeout, read ret: -1
8. exit tower
9. redo step 6 to 8 several times.
10. give up

11. Start Andropilot. A bluetooth connection appears on the bottom. Press button. A few moments later Andropilot is connected to the PixHawk....

ne0fhyk, 2015-02-12:
Tower might be trying to connect to a different saved Bluetooth device. In the settings screen, navigate to 'Advanced' -> 'Connection preferences' -> 'Bluetooth connection' -> 'forget Bluetooth device', and try to connect again.

A screen should pop-up with a list of Bluetooth devices to select from.

lvale, 2015-02-12:
@ne0fhyk Are you sure ??

A video is worth a thousand words

http://youtu.be/FE6p4RbrfCw

ne0fhyk, 2015-02-12:
@lvale yep, the `Connection Preferences` label can be clicked on:
[image]

[image]

lvale, 2015-02-12:
Great!!
On mine, as you can see it seems it is not an actionable option. Just got it right now.

closed 2015-02-12 by lvale

ne0fhyk, 2015-02-12:
@lvale great you got it to work! And thanks for the feedback!