DroidPlanner/Tower #1398: user camera profile not loading with tower

wolkstein, 2015-02-12:
hi,
my camera profiles (camera info xml files) which stored under /storage/emulated/0/DroidPlanner/CameraInfo are not available if i start Tower instead of DroidPlanner2. Is this a bug, or is the feature to create own camera info files in Tower simple not implemented?

wolkstein, 2015-02-15:
ok,
after some searching in repository i noticed that "/Android/src/org/droidplanner/android/utils/file/IO/CameraInfoReader.java" is removed and only available in branch dp2.
did this indicate that custom camera-profiles created by users are now completely removed from Droidplanner features list?

ne0fhyk, 2015-02-24:
@wolkstein the task of loading camera profiles is now handled by 3DR Services, hence the removal of the `CameraInfoReader` class.
Support for custom camera profiles will be added in the next release. 

ps: moving this issue to the [3DR Services issues list](https://github.com/DroidPlanner/3DRServices/issues/61).

closed 2015-02-24 by ne0fhyk