DroidPlanner/Tower #1399: Add Baidu Map

bys1123, 2015-02-12:
# Step One: Apply a key
### 1.1 Goto http://developer.baidu.com/map/index.php?title=%E9%A6%96%E9%A1%B5
[image]
### 1.2 Click “立即注册”
[image]
### 1.3 Fill the table (use gmail maybe fail)
[image]
### 1.4 Check your mailbox
[image]
### 1.5 Open the link in the mail that Baidu send to you
[image]
### 1.6 Wait few seconds you'll goto this page, sometime you'll goto the page in 1.1, that's ok, wait few minutes, try click the button described in 1.1 again.
[image]

### 1.7 Go to your mailbox as 1.5.
I remembered this gonna take two or three days. Baidu will verify the information. Or maybe in next step. 
### 1.8 Create one app
[image]

# Step Two: Replace the key in provider file
### 2.1 Our code is finish, but it's need to run some test. we'll pull it in two days.

phmagic, 2015-02-12:
hi @bys1123 i am stuck on the mobile number verification stage after I clicked on the email link.

bys1123, 2015-02-17:
hi @phmagic . Sorry for the late reply. 
The format is : "__(contry code)phone number__". For mobile of china, it's like (86)13810000000. In USA, I think it should be **(1)1234000000**. And the blue poptip on the right says "remove the 0 on the front of mobile number".
Sorry for the format in the picture, maybe let you thought 00 is International Code.
Try click "API控制台" on the top right corner of page 1.1, then you can back to page 1.6.

ne0fhyk, 2015-09-26:
@bys1123 was there any updates on this issue?

droneboost, 2016-07-14:
Hi All,

I've integrated Baidu map SDK in Tower in my own branch, here are some screen captures, I'd like to issue a pull request for code review. Is it OK?

**Source code:**
[image]

**Main view:**
[image]

**Map provider selection:**
[image]

**Map preference and offline map download:**
[image]
[image]

**Flight waypoints setting:**
[image]

ne0fhyk, 2016-07-14:
@droneboost yes that's great! Go ahead with the pull request.

droneboost, 2016-07-14:
@ne0fhyk, OK, I'll prepare the pull request.

onceme, 2016-07-14:
@droneboost Hey, man, you are the one today!

droneboost, 2016-07-28:
@ne0fhyk , I've sent the pull request https://github.com/DroidPlanner/Tower/pull/1777

would you please do the code review?

droneboost, 2016-07-28:
oops, seems tavis failed, I'll try to fix.

droneboost, 2016-08-01:
@ne0fhyk , new PR https://github.com/DroidPlanner/Tower/pull/1778

droneboost, 2016-08-13:
@ne0fhyk,
Shall we close this issue?

ne0fhyk, 2016-08-13:
@droneboost yes, let's close it.

closed 2016-08-13 by ne0fhyk