DroidPlanner/Tower #1405: Camera in Survey list: Sony Actioncams

jhellen, 2015-02-15:
Could someone help me with these two cameras? Was able to extract this info but I wasn't sure about sensor width and height. More info here http://www.sony.net/Products/di/en-us/products/me3d/specifications.html?contentsTop=1

<?xml version="1.0" encoding="utf-8"?>
<cameraInfo>
<Name>Sony HDR-AS100V</Name>
<SensorWidth></SensorWidth>
<SensorHeight></SensorHeight>
<SensorResolution>13.5</SensorResolution>
<FocalLength>17.0</FocalLength>
</cameraInfo>

<?xml version="1.0" encoding="utf-8"?>
<cameraInfo>
<Name>Sony HDR-AS30V</Name>
<SensorWidth></SensorWidth>
<SensorHeight></SensorHeight>
<SensorResolution>11.9</SensorResolution>
<FocalLength>15.3</FocalLength>
</cameraInfo>

jhellen, 2015-02-15:
Should these be added as 4 different cameras because of the different angles caused by SteadyShot?
170° (without SteadyShot),
120°(with SteadyShot)

ne0fhyk, 2015-09-26:
@jhellen any updates here?

ne0fhyk, 2016-08-17:
Closing for lack of updates.

closed 2016-08-17 by ne0fhyk