DroidPlanner/Tower #1406: Channel 8 Problem with Tower

Darrinmu, 2015-02-15:
I have channel 8 set for my retractable legs.  Yesterday, when I connected to Tower for the first time, the legs retracted.  In order to get them to un-retract, I had to unplug my quad and re-connect.  This time, I successfully retracted legs in flight, but they would not un-retract.  I recreated the issue at home multiple times.  Channel 8 would work when the quad was initially plugged in and Tower was connected, but as soon as I changed modes (Stability, Loiter, Auto), the legs would stop working.  

I turned off Tower and turned on DroidPlanner.  I was able to connect and to raise and lower my legs on channel 8 and to switch flight modes with DP with no issues.

Bradatech, 2015-02-17:
This sounds like it is related to our issue https://github.com/DroidPlanner/droidplanner/issues/1394 since Tower appears to mess with the RC outputs on the Pixhawk upon mode changes.

Darrinmu, 2015-02-18:
Thanks, I will check out that issue.  I solved some of my problem by disabling the "Pan" action in Mission Planner.  Tower and DP were trying to operate the Pan capability, but that signal was actually tied to my retractable legs, not Pan.  I had a successful flight today with DP.  I will try Tower again, but I am still not sure why it started activating the channels as soon as it connected to the copter.

ne0fhyk, 2015-02-24:
@Darrinmu @Bradatech the issue you encountered was caused by a bug in `3DR Services` and has been fixed in `version 1.2.9`(https://github.com/DroidPlanner/3DRServices/pull/63).

That version is currently in the beta channel for testing, and will be released on the play store by the end of the week.

closed 2015-02-24 by ne0fhyk