DroidPlanner/Tower #1417: feature request

stixchicken, 2015-02-25:
Add altitude selection to takeoff also  need to be able to select diffrent bluetooth devices once you select a bluetooth device the only way to select a diffrent one is to uninstall tower and reinstall

ne0fhyk, 2015-02-25:
@stixchicken you can switch Bluetooth devices by going into the settings to 'Advanced' -> 'Connection preferences'.

stixchicken, 2015-02-25:
Ok i found another issue then you cant get to the advanced menu on an lg3

ne0fhyk, 2015-02-25:
Could you be more specific or provide screenshots of the issue. 

In the settings screen, taping on the 'Advanced' label should take you to the advanced screen. 
On that screen, tap on the 'Connection Preferences' label to access that section. There, you should be able to remove the saved Bluetooth device.

stixchicken, 2015-02-25:
Yea i found that thanks but what does the advanced menu checkmark do if i check it itcdosentcseem to change anything

closed 2015-04-23 by ne0fhyk