DroidPlanner/Tower #1420: Add an option for custom camera setup

nhadrian, 2015-02-28:
Hi,

if it is possible, please add the possibility of configuring custom camera setups for surveys.
For example I'm using Canon IXUS100 camera either for mapping and another piece modified for NDVI. I know any details about them (sensor dimensions, resolution, focal, etc…)

Thanks in advance.

Regards
Adrian

wolkstein, 2015-03-04:
hi,
i am not one of the dev team. but custom camera profiles are possible.
you can create an own camera.xml file and must place them into the droidplanner or tower folder.

for tower this is:
/storage/emulated/0/Tower/CameraInfo
and for older droidplanner2 this is:
/storage/emulated/0/DroidPlanner/CameraInfo

example file for gh4 with 14mm focal length and 16:9:
```
<?xml version="1.0" encoding="utf-8"?>
<cameraInfo>
  <Name>GH4_14mm_16:9</Name>
  <SensorWidth>17.3</SensorWidth>
  <SensorHeight>9.73125</SensorHeight>
  <SensorResolution>11.9</SensorResolution>
  <FocalLength>14</FocalLength>
</cameraInfo>
```
you can name the file as you like. for example Canon_IXUS100.xml . the file extension "xml" is needed.

after this the new camera profile is listed on survey or structure scanner widget.

nhadrian, 2015-03-04:
Hi,

great, thanks!
I'll follow this way.

Regards
Adrian

PS.: maybe issue could be closed then?

ne0fhyk, 2015-03-04:
@wolkstein thanks for the response. Quick clarification though, as highlighted in https://github.com/DroidPlanner/3DRServices/issues/61, the path to place the custom files for Tower should be `/sdcard/3DRServices/CameraInfo/`

closed 2015-03-04 by ne0fhyk

GeVaSta, 2015-04-24:
This path does not exist on my Nexus 7. No folder under sdcard named 3DRServices.

ne0fhyk, 2015-04-24:
@GeVaSta that directory is not created automatically. You can created manually and add the needed files there, and restart Tower and 3DR Services.
Once restarted, the app will detect the directory you created, and the data inside.

GeVaSta, 2015-04-25:
Right. I'll give it a try. Thanks.

burkeam, 2015-09-18:
I'm not able to get a new camera profile to take. Please help if able.

The history
I've tried everything I have found but cannot add a new camera to tower.

I've been trying to add a new camera Tower and can't get anything to work. Tried in 4 different location adding the CameraInfo/Mapir_Camera.xml. Force stopped 3dr service, tower and solo. Tried rebooting. Nothing. 

Has something changed?

The Tower app show the storage at: Which I cannot change 
/storage/emulated/0/Tower/ 

I'm using a S5 Sport
I created the following folder structures and file in 4 different places 
storage/emulated/0/3DRServices/CameraInfo/Mapir_Camera.xml
storage/emulated/0/Tower/CameraInfo/Mapir_Camera.xml
storage/extSdCard/3DRServices/CameraInfo/Mapir_Camera.xml
storage/extSdCard/Tower/CameraInfo/Mapir_Camera.xml

Here's the text in the .xml
[image]

Any ideas

Jubalr, 2015-10-18:
@GeVaSta did you ever get this to work? I have the Nexus 7 as well and created the folders as shown, but camera still does not show up.

GeVaSta, 2015-10-19:
@Jubalr
Not yet because I did not really try it.

ramirezmaps, 2016-05-25:
i need help to install new camera in tower, procedure in not work!!