DroidPlanner/Tower #1428: Add functionality to cancel ROI and resume heading in direction of travel

idleup, 2015-03-06:
Currently it will face the ROI until either the mission is completed or you define another ROI. There are times you along want to face an ROI between specific waypoints and then the duration of the mission have it face direction of travel. I have not found a way to do this currently.

PaulLoizeaux, 2015-03-09:
To reset the ROI to the direction of travel just insert a second ROI at the point you want it to revert to normal and have Lat, Lon, and Alt be 0.

Source: http://copter.ardupilot.com/wiki/mission-planning-and-analysis/mission-command-list/#do-set-roi

Thalek, 2015-03-09:
Sounds rather non-intuitive, to me.

ne0fhyk, 2015-03-09:
@Thalek the plan is to add a 'cancel ROI mode' to the ROI waypoint, that will hide/abstract the reset mechanism.

Thalek, 2015-03-10:
Ahhhh.  Sounds good.

ne0fhyk, 2015-09-26:
duplicate of #1267

closed 2015-09-26 by ne0fhyk