DroidPlanner/Tower #1430: UI Feature Request: a visible method of distinguishing multi-function buttons.

Thalek, 2015-03-07:
Per a conversation here:  https://groups.google.com/forum/#!topic/droidplanner/M8iRCk32iu4

Right now, there are buttons that appear to require only a single press to activate their function.  There are also buttons that activate additional features with a long press.  And there are still others that activate additional features with a double-tap.  And they all look exactly alike, and sometimes, have little in the way of documentation to explain their use or where to find a given feature.  (Correction: four types, as some of the number functions can either be changed by sliding a finger, or you can do a long press of the button to manually enter a number via keyboard.)

It is my suggestion that the various types of button have some way of being visually distinguished.  Perhaps a double-tap button would include a plus sign ("+") as part of the button's graphic, and all double-tap buttons would have this.  A long press button might have, say, a wavy border, and all long-press buttons would have this wavy border.  A slide button might have directional arrows added to the graphic.  And a button that has no additional "hidden" functions wouldn't need to be changed at all.

So, if you had a button that could be double-tapped, slid, AND a long press to access different additional features, it would have all three graphical changes.  Obviously, the changes would have to be consistent, visually distinctive, but still small enough not to overshadow the graphic already existing on the button.  No easy task, I understand.

Unfortunately, I am experiencing some health issues, so I have not roughed out any art for this yet. Hopefully, I would be able to take some screen captures and make modifications over the weekend.

ne0fhyk, 2015-03-09:
@Thalek thanks for adding the issue, and for the great description! Hope you get better soon!

@zumbrunnen any input on the ux for this issue would be appreciated!

Thalek, 2015-03-09:
@ne0fhyk glad to be of service.  I'll have to consult my doctor on the health issues, though. It might be time to change anti-depressants.  Again.

ne0fhyk, 2015-09-26:
The new editor UI in version 3.2.0 should fix this issue.

closed 2015-09-26 by ne0fhyk