DroidPlanner/Tower #1432: Add virtual joystick support

ne0fhyk, 2015-03-09:
More details to be added...

Loop-Zilla, 2015-03-10:
ne0fhyk, 

I agree.  At least bring back what was commented out in DroidPlanner.  If this feature is too risky for aerial assets, bring it in only for rovers and/or gimbal control.  I don't use an RC Tx anymore with our rovers, but the ability to back out of a corner with the virtual joysticks would be great.

Linjieqiang, 2015-03-11:
great!

rmackay9, 2015-03-11:
I would really like it if we didn't use rc-overrides for this but instead used one of these alternative mavlink messages:
http://mavlink.org/messages/common#SET_ATTITUDE_TARGET
http://mavlink.org/messages/common#SET_POSITION_TARGET_LOCAL_NED
This would require a little work on the arducopter side to grab and interpret these messages while in Guided mode but longer-term I think it would be simpler and easier because it avoids muddling RC controlled modes (Stabilize, Loiter, etc) and tablet controlled modes (Guided, Auto).

ne0fhyk, 2015-03-11:
@rmackay9 That's the option Chris and I are leaning toward since it avoids the possible issues associated with RC Overrides.
Along the messages you mentioned, couldn't [`SET_POSITION_TARGET_GLOBAL_INT`](http://mavlink.org/messages/common#SET_POSITION_TARGET_GLOBAL_INT) be used as well?

billbonney, 2015-03-11:
You might want to consider this message as well http://mavlink.org/messages/common#MANUAL_CONTROL
Not only does it offer the x,y,z,r controls, it offers a set of 16 push buttons as related to a joystick

R-Lefebvre, 2015-03-11:
That would be great, as it would also move us on another issue which is to enable Aux Functions from the tablet.

rimb05, 2015-05-26:
I think having the ability to touch a virtual joystick to deviate from a guided or auto course would be great. When the user lets go of the joystick, guided/auto mode should resume.   I don't think there's much risk as long as joystick mode works like FBWB while its flying under joystick control. 

If possible, having the ability to only activate one axis of the joystick (horizontal, for example) would make it very safe, as it would only allow course deviations.

ne0fhyk, 2015-09-26:
Replaced by #1518

closed 2015-09-26 by ne0fhyk