DroidPlanner/Tower #1433: Add a 'Kill Switch'

ne0fhyk, 2015-03-09:
Similar to Parrot's ar-drone/bebop emergency mode, the kill switch should allow the pilot to immediately disable (disarm) a connected copter.

Loop-Zilla, 2015-03-09:
Second that request.

Thalek, 2015-03-09:
Just require a double-tap or a long tap to make it harder to activate purely by accident.

zlite, 2015-03-09:
I just dropped a BeBop from 40 feet by accidentally hitting the Emergency Mode.  Fortunately it's very robust (because it's small and light), but for anything larger we'll need a pop-up to ask if you're sure.

Loop-Zilla, 2015-03-09:
I would love to have a hardware E-Stop implementation, maybe even something through the headphone jack (not sure how that works, but I have seen devices use the port for analog sensing).

BitMapping, 2015-03-21:
Somewhere Rob suggested a slider. It is more secure compared to most other inputs.

ne0fhyk, 2015-04-24:
Fixed by PR #1470

ne0fhyk, 2015-04-24:
Disabling the feature until AC 3.3 is released since it adds support for `disarm` while in flight.

ne0fhyk, 2015-05-02:
For versions prior to AC 3.3, starting accelerometer calibration could be used as a kill-switch, since it'll cause the autopilot to block until calibration is complete.

More of a hack though!

squilter, 2015-07-07:
For versions before 3.3, the kill switch will change the mode to `stabilize` and then disarm.
For 3.3+, the kill switch will send the disarm packet with the the force flag (#1494).

squilter, 2015-07-08:
@ne0fhyk How is the arducopter version checker going?

ne0fhyk, 2015-07-09:
Fixed by #1511

closed 2015-07-09 by ne0fhyk

MaxQu, 2015-08-11:
Support the emergency disarm. Will save blade when the drone tips over.