DroidPlanner/Tower #1434: Reset / Restart Mission and 'Go to WP...'

Loop-Zilla, 2015-03-10:
I have searched through Tower extensively for this option, and really can't believe it isn't in there.  Is the accepted method simply to re-upload the mission?  With a lot of waypoints, this is cumbersome.

It would be a nice feature to be able to restart or reset the mission, not just recover and try to accomplish the last known WP destination.  I experience this need more frequently with rovers, not so much with aerial assets.  However, there are times when I launch an aircraft, go to manual for some issue (eg. other air traffic), then want to simply restart the mission.

Additionally, it would be handy to tell the asset to go directly from current position to another WP and continue the mission (or hold, or whatever).

Loop-Zilla, 2015-03-11:
Driving around last night, I aborted a mission by hitting hold and then RTL.  I edited a new mission, and uploaded the waypoints.  When I switched to Auto for the 'new' mission, the AP tried to go back to the 'old' mission's last achieved waypoint.  

I cleared the mission in Tower, and uploaded no waypoints to try to clear it out.  Then, created yet another new mission and uploaded it.  Switch to Auto, and it still headed to the original 'old' mission's last achieved waypoint.

I let the asset drive it out to the end.  Created and uploaded yet another 'new' mission, went to Auto, same thing.  Only a reset of the PixHawk would clear it out and let me conduct a new mission.  

Is this a Tower or ArduRover firmware bug?  I'm glad I'm figuring this out with a rover, in a large parking lot with no obstacles...

ne0fhyk, 2015-03-11:
That seems to be an ArduRover bug. I experienced the same behavior using
the emulator (SITL) with the latest ArduRover firmware.
That behavior is not reproducible with either ArduCopter or ArduPlane.

squilter, 2015-07-06:
@Loop-Zilla This is specific to ArduRover.  It does not occur with Copter or Plane.  The solution is to set `MIS_RESTART` to 1.

[Link to parameter docs](http://rover.ardupilot.com/wiki/apmrover2-parameters/#mission_restart_when_entering_auto_mode_mis_restart)

closed 2015-07-06 by squilter