DroidPlanner/Tower #1437: support EKF_STATUS_REPORT message

rmackay9, 2015-03-12:
AC3.3 provides support for an EKF_STATUS_REPORT message.

The report has a "flags" field and a list of variances which provide details on how good the EKF thinks it's estimates are for various things like horizontal-position and compass.

From a previous discussion with Paul, we think the GCS should have an EKF details page or maybe just a little EKF icon which is red, amber or green.  The GCS could look at all variances, and decide on the EKF icon's colour by taking the worst variance of the list and: 
               Below 0.5 = good (i.e. green)
               0.5 ~ <0.8 = warning  (i.e. amber)
               0.8 or higher = bad (maybe trigger failsafe) (i.e. red)

    
    <message name="EKF_STATUS_REPORT" id="193">
        <description>EKF Status message including flags and variances</description>
        <field name="flags" type="uint16_t">Flags</field>   
        <field name="velocity" type="float">Velocity</field>
        <field name="pos_horiz" type="float">Horizontal Position</field>
        <field name="pos_vert" type="float">Vertical Position</field>
        <field name="compass" type="float">Compass</field>
        <field name="terrain_alt" type="float">Terrain Altitude</field>
        <field name="unused1" type="float">Unused1</field>
        <field name="unused2" type="float">Unused2</field>
        <field name="unused3" type="float">Unused3</field>
    </message>
 
     The flags message mirror’s our getFilterStatus() output.  I’m wondering if we could remove the *PRED* enums?
 
                
    <enum name="EKF_STATUS_FLAGS">
        <description>Flags in EKF_STATUS message</description>
        <entry name="EKF_ATTITUDE" value="1"> <description>set if EKF's attitude estimate is good</description></entry>
        <entry name="EKF_VELOCITY_HORIZ" value="2"> <description>set if EKF's horizontal velocity estimate is good</description></entry>
        <entry name="EKF_VELOCITY_VERT" value="4"> <description>set if EKF's vertical velocity estimate is good</description></entry>
        <entry name="EKF_POS_HORIZ_REL" value="8"> <description>set if EKF's horizontal position (relative) estimate is good</description></entry>
        <entry name="EKF_POS_HORIZ_ABS" value="16"> <description>set if EKF's horizontal position (absolute) estimate is good</description></entry>
        <entry name="EKF_POS_VERT_ABS" value="32"> <description>set if EKF's vertical position (absolute) estimate is good</description></entry>
        <entry name="EKF_POS_VERT_AGL" value="64"> <description>set if EKF's vertical position (above ground) estimate is good</description></entry>
        <entry name="EKF_CONST_POS_MODE" value="128"> <description>EKF is in constant position mode and does not know it's absolute or relative position</description></entry>
        <entry name="EKF_PRED_POS_HORIZ_POS_REL" value="256"> <description>set if EKF's predicted horizontal position (relative) estimate is good</description></entry>
        <entry name="EKF_PRED_POS_HORIZ_POS_ABS" value="512"> <description>set if EKF's predicted horizontal position (absolute) estimate is good</description></entry>
                </enum>

rmackay9, 2015-04-27:
We're about 1 month from the AC3.3 release now.  We will definitely want this message supported for that release and providing it earlier so that it can be tested would be greatly appreciated.

ne0fhyk, 2015-08-31:
Fixed by pr #1583

closed 2015-08-31 by ne0fhyk

rmackay9, 2015-08-31:
yeah!  thanks!