DroidPlanner/Tower #1441: problem with flight time

arnaudstephan, 2015-03-17:
Flight time don't change when modifing the speed by adding a change speed point in a mission.

Other question : what is the speed of a survey ?

Thanks

ne0fhyk, 2015-03-17:
the speed of a survey is the speed defined by the [`WPNAV_SPEED`](http://copter.ardupilot.com/wiki/configuration/arducopter-parameters/#waypoint_horizontal_speed_target_wpnav_speed) parameter. However, if you add a change speed point in a mission before the survey, then the speed will be temporarily set by that point.

arnaudstephan, 2015-03-17:
Thank you. And what about the wrong flight time calculation ?

ne0fhyk, 2015-03-17:
That's an issue indeed. Right now the time calculation is estimated based on the WPNAV_SPEED parameter only. 
That needs to be fixed to account for any change speed point inside the mission.

ne0fhyk, 2016-09-25:
Fixed by pr #1815

closed 2016-09-25 by ne0fhyk