DroidPlanner/Tower #1445: Battery capacity estimate based on voltage

BitMapping, 2015-03-21:
If a battery is not fully charged Tower reports 100% capacity. The reason is clear. 
To avoid confusion it might be helpful if Tower would guesstimate the remaining capacity based on the voltage and the battery capacity in the params using some simple formula.

Linjieqiang, 2015-03-27:
I found the same problem. I think you are right!

zill539, 2015-04-23:
I'm with you on this one. I don't always fly with a full pack and therefore i always have to keep an eye on the voltage, all that's fine but that renders  the capacity meter useless. Cbased on voltage would be a nice feature and maybe have the option to turn off the capacity display wouldn't be a bad thing.

squilter, 2015-04-28:
I think this one is better suited for flight code.  Let those guys know that there is interest for this feature by commenting on diydrones/ardupilot#1434