DroidPlanner/Tower #1447: WP_NAV_SPEED slider in survey menu

BitMapping, 2015-03-21:
Would be good to have WP_NAV_SPEED adjustable in the survey dialog. Otherwise it has to be adjusted in the Parameters which is at least clumsy (or impossible when using xBee telemetry).

ne0fhyk, 2015-09-26:
A `change speed` mission item was added that allows to change the speed of the vehicle during a mission.

ThorstenDP, 2015-10-03:
Ok, I will check that.
However, I think that it would be much easier as well as powerful to implement it in the survey dialog. Or are there any reasons arguing against it?

ne0fhyk, 2015-10-03:
No technical reasons against it. Beneath the scenes, it can generate a change speed mission item, but I guess the ux change would make things easier for users?

ThorstenDP, 2015-10-03:
I am often flying subsequent missions at different altitudes. And for planning the correct overlap - i.e. not flying to fast and thus miss some images due to a too slow camera or to increase speed for higher altitudes - this would be a great feature. It might be done manually using the change speed item but another advantage of including it in the survey menu would be that the (average) camera trigger interval in seconds can be calculated for direct verification of all relevant mission parameters.

SciAero, 2015-10-28:
Agreed this should be added to the survey options so that the change speed waypoint is put at the first waypoint, but also another change speed waypoint should be put at the end of the survey to go back to default speed for moving to another mission or returning home.