DroidPlanner/Tower #1452: Feature request : Change Telemetry ID

oocciinn, 2015-03-31:
Having the ability to change the Telemetry radio ID would be useful feature in certain instances, in my case when testing multiple vehicles this would allow me to bring only one ground radio to the field.

PaulHam211, 2015-04-01:
Why not set all of YOUR craft to the same net ID, I wouldn't have thought you are able to fly more than 1 at a time :).
Although I do agree that this would be a good addition to the firmware as you may find that someone at your flying field is also running the same net ID as you and you need to change it.

Thalek, 2015-04-01:
I would think this would be useful for when you are flying multiple craft in a swarm, such as, say, doing an aerial survey.