DroidPlanner/Tower #1459: Splined missions to avoid jerks at waypoints

ThorstenDP, 2015-04-16:
Hi all,

it would be great if there is an option to execute survey missions based on spline waypoints. This would avoid the nasty jerks at the waypoints.

Cheers,
Thorsten

PS: I hope that the development of Tower is not abandoned after the release of the Solo and splitting up droidplanner into DroneKit-Android and Tower.

ne0fhyk, 2015-04-18:
Hi Thorsten,

Tower development will continue, and it'll even be possible to control Solo
with Tower.

ThorstenDP, 2015-04-18:
Hi Fredia,

great news! I was scared somehow, because Tower is an awesome app but development slowed down for quite a while. I used it for hundreds of survey missions and hope that the survey section will receive some more attention in the future - even though it is currently not 3DR's major focus.
Keep up the good work!

Thanks,
Thorsten

squilter, 2015-07-21:
Fixed by #1522

closed 2015-07-21 by squilter