DroidPlanner/Tower #1460: Request: convert survey to waypoints

chapman, 2015-04-20:
The survey tool is really slick, but sometimes you want to modify a few waypoints afterwards, and there's no way I can see to do that within the app, right? Another use case is just to check the mission before flying it, if you want to see how the survey tool translates into waypoints (e.g. is it one per leg, or multiple)

One hack is to upload the mission to the UAV and then download back to the app, but that's a bit messy and you need to be connected at the time, so not so good for pre-planning.

In short, it would be good to click on a survey, hit 'Convert to Waypoints' and see it in the app how the UAV will see it

ThorstenDP, 2015-04-21:
I second that! It was possible in DP1 and I used it quite a lot.

ne0fhyk, 2015-08-05:
@TShapinsky similar to what's already been done. Just add the option to the complex mission items detail window to trigger it manually.