DroidPlanner/Tower #1462: Request: dragging coincident points chooses most recent one

chapman, 2015-04-20:
There's a really frustrating workflow if you drop a new waypoint over the top of an existing one (which seems too easy to do, perhaps a preference for this spacing threshold?)

- drop new waypoint over the top of another
- hold+drag drags the old one, which you don't want to move, but you have to move that away
- then you can hold+drag on the new one, move it where it should be
- go back to reset the position of the older one

This would be alleviated if the newest waypoint is on 'top' and will take precedence in a coincident point situation

squilter, 2015-04-21:
I think this is related to both #1079 and #952.  Ultimately, we concluded there was no proper fix for either.

ne0fhyk, 2015-04-23:
@chapman as @squilter pointed out, we're working within the limitations of the google map api, and there's no proper fix for the issue you encountered.

closed 2015-04-23 by ne0fhyk