DroidPlanner/Tower #1463: allow user to set-home and prompt for this if home is captured in-flight

rmackay9, 2015-04-22:
It is possible for users to take-off without first getting a GPS lock and valid home position.  In this case copter will record the home position while the vehicle is flying.  This can lead to the home position being quite far from the actual home position or over a location where it's not safe to land.

It would be good if the GCS did this:
 - watch for the home position being set while the vehicle is in flight.  The vehicle should send a WP#0 when this happens although the GCS will need to be careful not to confuse this message with a WP#0 received in response to a mission download.
 - if the home position is set during flight the GCS should update the home location on the map and give a verbal warning (optional) to the pilot to say the home has been set and request they move it to a more accurate position
 - allow the pilot to home the home icon around (optionally providing a suggested location using the tablet/computer's GPS) and once set, send a do-set-home command to the vehicle (optionally using the terrain altitude at that horizontal location).
Now when the vehicle RTLs it will RTL to the pilot provided home location.

hamishwillee, 2015-04-22:
Above sounds good. Could the initial default home position instead be based on the GCS GPS position (in some way) if not obtainable from the device. At least you know it is likely to be land, and "about right".

ne0fhyk, 2015-09-26:
Fixed by pr #1611 && https://github.com/dronekit/dronekit-android/pull/285

closed 2015-09-26 by ne0fhyk