DroidPlanner/Tower #1464: When using auto-pan; zoomlevel gets wrong; Zoombuttons missing.

GeVaSta, 2015-04-22:
When you use the autopan feauture (touch and hold the location icon), after a disconnect or perhaps even GPS loss, the zoomlevel changes to displaying the whole world.
It is then very difiicult to zoom in onl by pinching a lot off time. 
A zoom + and - button would be preferable too.

ne0fhyk, 2015-04-23:
Thanks for the feedback, but there's no plan to add zoom buttons, as they would clutter the ui.

closed 2015-04-23 by ne0fhyk

GeVaSta, 2015-04-23:
Have you already fixed te spontaneous zoom-out? Because you closed this case I suppose you did. Or did you misunderstand my issue?
Solving the zoom-out on disconnect issue, makes the need for zoombutton less important.

ne0fhyk, 2015-04-23:
I'm not able to reproduce the zoom out issue you're mentioning. The map
view should stay on the same zoom level when the user disconnects, or when
the link between the GCS and vehicle breaks.

sditges, 2015-04-23:
i have the same trouble with zoomout

GeVaSta, 2015-04-23:
I'll try to find out when exactly it is happening and report back.