DroidPlanner/Tower #1473: Option to set WP_YAW_BEHAVIOR = 0 ("Never change yaw")

ThorstenDP, 2015-04-27:
Hi all, 

I want to bring this up again. It was discussed but got lost in some closed issues.
(https://github.com/DroidPlanner/Tower/issues/884
https://github.com/DroidPlanner/Tower/issues/325)

I talked to some other UAV pilots using other Flight controllers and all chose something like setting WP_YAW_BEHAVIOR to 0 ("Never change yaw").

There are several advantages:
- Security: you always know the orientation of the copter in case you have to fly manually 
- Easy image analysis: scanning through the images visually it is much intuitive if they all show the same orientation
- Saving battery and time: It should be (at least a little bit) more efficient if any additional movement is avoided.

The main objective is security. Knowing the orientation is especially important when the mission covers a larger area and you operate at the limits of line of sight. If the copter is 300+ m away it is very hard to see if it turns, how fast it is and if it is still flying away from you or already coming back to you. Keeping track of the copter is crucial. If you become distracted for just a second it is sometimes hard to find it again - not to mention its flight direction. It also strongly depends on your position relative to the survey grid, as well as the distance between you and the copter. This is for sure no problem for small area missions, but it is for any grid larger than 250*250m. If yaw is fixed you always know the orientation of the copter. This is why I never change it. It might be a life saver. 
Would be great if you could give it a try.

Maybe the easiest way implementing it would be:
- take off
- go to WP1
- set WP_YAW_BEHAVOR = 0
- set new yaw (to hatch angle)
- set CAM_TRIGG_DIST > 0
- proceed mission
- RTL

Thanks and cheers,
Thorsten

ne0fhyk, 2016-08-19:
Fixed by commit 14f6fc2b816ee3e64de0da9d67a8d22480da3da5

closed 2016-08-19 by ne0fhyk

BitMapping, 2016-08-19:
@ne0fhyk 
Hi Fredia,

thanks a lot for implementing this! Really appreciated!
I am looking forward to testing it. What is the schedule for 4.0?

One more thing that would be really great to have it implemented is #1447 to set WPNAV_SPEED in the survey dialog. I think then it is fully featured.

Cheers and thanks again!
Thorsten

ne0fhyk, 2016-08-22:
@BitMapping a beta for 4.0 should be available later this week. The speed slider setting is likely to land in the next release though (v4.1.0)