DroidPlanner/Tower #1474: Effective Interleaved Survey II

ThorstenDP, 2015-04-27:
Hi all,

I would like to reopen https://github.com/DroidPlanner/Tower/issues/637 again since it was not closed because it is not possible or not reasonable.
Following my suggestions on survey planing (#325) I have some idea to make a survey more effective in terms of flight time.
What reduces endurance in a grid survey design is that the copter slows down when reaching a waypoint and then has to speed up again. This takes time and also drains the batteries.
To minimize this effect I was thinking of some "phase shift" (interleaved, see Arthur's first comment on the original issue, https://github.com/DroidPlanner/Tower/issues/637) approach to grid surveys as shown in the figure. A is the common approach, B the phase shift concept. 
The disadvantage of B is that the flight path needs to be a little longer to cover the same area. However, this should be compensated, because the first and last waypoint are neighbouring and thus the flight path overhead (dashed line in A) is smaller (at least in this design).
Maybe this is all too complicated in terms of optimizing line spacing, radii, navigation speed, etc. but I wanted to share the idea anyway to start a discussion on making surveys more effective.

[image]

A combination with splined surveys (https://github.com/DroidPlanner/Tower/issues/1459) would be optimal.

Thanks again!
Thorsten

ThorstenDP, 2015-04-27:
This would also help minimizing effects/artifacts in orthomosaics due changes in illumination during a flight.

FLYBYME, 2016-05-09:
Think we will ever get something like this?

ne0fhyk, 2016-05-11:
@ThorstenDP  @FLYBYME I'm just catching up on the conversation. What's the advantage of the proposal over the currently available splined surveys?

ThorstenDP, 2016-05-11:
@ne0fhyk 
I think the biggest advantage is that the first WP and the last WP are very close together. This would be a major gain in efficiency and security as well. In common missions the battery gets lower the further the copter is away from the pilot (Well it depends on where the fist WP is located in relation to the pilot*). But in there interleaved case the copter still has lots of battery when over that area.
Additionally, as mentioned above the copter slows down when reaching a waypoint and then has to speed up again. This takes time and also drains the batteries. 
Not sure if Arthur's old implementation still fits into the new code base... But I would really appreciate such a feature.

*An additional important feature for grid surveys would be if one could specifically set the location of the fist WP as the starting point for the grid. It would also be good if the order would be shown in the Editor - at least the location of the first and last WP. Maybe the option to reverse the order of the WPs would be sufficient. At least it would make mission planning much more powerful.

ThorstenDP, 2016-05-16:
@ne0fhyk 
The following image shows the exposure (blue: good exposure, red: bad exposure) of the single images of a grid survey mission.

[image]

This is probably not the best example but it shows some regions with exposure problems stemming from changes in illumination - it was partly cloudy. This results in orthomosiacs showing trends in luminance and color. It could partly be solved with interleaved mapping since neighboring flight paths are not surveyed in direct sequence. As a result exposure compensation would be easier.