DroidPlanner/Tower #1481: Support multiple vehicles and components

rmackay9, 2015-05-05:
I thought I raised this issue before but I can't find it on the list.

DroidPlanner gets confused when there's multiple vehicles or multiple components sending a heart-beat (and perhaps other) messages.  The most obvious issue is when a SToRM32 gimbal is connected and it results in the flight mode changing back-and-forth from the vehicle's actual flight mode to "UNKNOWN".

I know it's a big ask but it would also be nice if DroidPlanner supported viewing and controlling multiple vehicles which is now possible with AC3.3 (and even AC3.2.1) as shown in this blog&video:
http://diydrones.com/profiles/blogs/multi-vehicle-testing-with-apm-copter-tracker-and-mission-planner

ne0fhyk, 2015-10-11:
@visper1 which version of Tower and 3DR Services are you using. The latest version should filter out sys and component id that don't match the vehicle's.

@visper1 @rmackay9 would you know what is the sys and component id for the antenna tracker.

visper1, 2015-10-11:
@ne0fhyk I using Tower 3.2.0, Pixhawk with last fw 3.3 and last fw APM antenna tracker from here: http://antennatracker.ardupilot.com/wiki/loading-the-firmware/

rmackay9, 2015-10-12:
@ne0fhyk, the sysid seems to be "2" and component id "1" for the tracker by default.

arsenixprime, 2016-05-13:
As of current Tower-V3.2.1-Beta this still doesn't work. Has this been pushed to Tower v3.2.2?

ne0fhyk, 2016-05-13:
@arsenixprime yes this has been pushed to the next release.