DroidPlanner/Tower #1482: cannot import my old missions into tower

wolkstein, 2015-05-11:
hi,
i have missions in normal txt format which i will import into tower. but no way, tower allows only to open dpwp waypoints which also looks binary. while trying to open an waipoint dpwp file, my text editor (kate) show me an binary file. how can i convert my important (old) missions to dpwp files?

/g
wolke

mutatron, 2015-07-07:
I figured out a workaround. It's a pain in the ass if you have a lot of them, though. I used the Pixhawk to do the conversion for me:

1. Hook up receiver to MacBook
2. Connect to Pixhawk with Mavlink in APM
3. Load waypoint mission file in APM
4. Upload waypoints to Pixhawk
5. Move receiver to Nexus 7 with Tower running
6. Connect Tower to Pixhawk
7. Download Mission
8. Export to File
9. If more files, go to 1

ne0fhyk, 2016-08-17:
Fixed by pr #1791

closed 2016-08-17 by ne0fhyk