DroidPlanner/Tower #1483: APM is not interpreting params as per the MAVLink standard for non-float values.

kd0aij, 2015-05-13:
this results in corrupted parameter values when working with the PX4 native stack

LorenzMeier, 2015-05-13:
The issue is the param Union value: it should contain the binary representation of the parameter, while it always contains the float value for APM. This leads to rounding errors on integer parameters.

ne0fhyk, 2015-05-13:
According to the mavlink definition, the param value type is a float (
https://pixhawk.ethz.ch/mavlink/#PARAM_VALUE).

Or are you referring to another mavlink message?

LorenzMeier, 2015-05-13:
I wrote the original spec, so I'm quite confident about its interpretation. I can see that it should have been more bold about the way the float type has to be interpreted. The same way as the waypoint protocol spec is not covered by just reading the XML file, the parameter protocol is not covered by just reading the XML file.

The MAVLink spec is here:
http://qgroundcontrol.org/mavlink/parameter_protocol
(linked to by http://mavlink.org)

The problem is that the APM implementation wasn't done by reading the specs. If you think about it for a moment the MAVLink message makes no sense when always transmitting the parameter as float: Why would you provide a parameter type enum if the value itself is *always* float?

kd0aij, 2015-05-21:
This issue makes Tower unsafe for use with PX4...

kd0aij, 2017-02-20:
closed assuming Tower is no longer being maintained

billbonney, 2017-02-20:
@kd0aij It is still being maintained... It seems no one added PX4 support as of yet..  This issue is still open until is fixed. Thx.