DroidPlanner/Tower #1486: Add option to change default height / altitude

gmaclennan, 2015-05-21:
Tower currently defaults every waypoint to 20m. I cannot find any way to change that other than changing each waypoint manually. Would be great to be able to change the default height for new waypoints. We fly in rainforest regions where 20m would not clear the trees and one mistake would result in a dead drone :-(

ne0fhyk, 2015-05-22:
@gmaclennan you can either set a `takeoff` waypoint, and specify the takeoff altitude which will cause the waypoints added afterward to use it as their base altitude.
Alternatively, you can also select all the waypoints you'd like to update via the `selector` tool, and open the right side detail panel to edit all the selected waypoints in bulk.

[image]
[image]

closed 2015-05-22 by ne0fhyk

gmaclennan, 2015-05-22:
Thanks @ne0fhyk this is very helpful.

It would still be useful to define a default though, since the current default is kind of arbitrary, and changing the height every time is an additional step for users.

PR #1489 (2015-05-26): Feature custom altitude preferences
Add the ability to customize the default altitude, as well as the min and max altitude offered in the app.

The preferences can be found under `Settings` -> `Advanced` -> `Altitude Preferences`.

Also fixes #1486 , #1244