DroidPlanner/Tower #1487: "Connection failed"  (3DR Radio)

AndKe, 2015-05-23:
I've reproduced this on two android devices, with two different radios.
(Radios are properly configured and works with laptop+APM Planner, solid green LED)
Tower is current version from Google Play.

Pressing "connect" takes some time , the "connection failed" appears.  
"Advanced setup" confirms baud rate is 57600.

ne0fhyk, 2015-05-24:
@AndKe Make sure 3DR Services has permissions to access the usb connected radios. If you have Droidplanner 2 installed, it may be that DP2 is set as the default app to gain usb access permissions.

AndKe, 2015-05-24:
Good point, but I always used the "this time only" option when selecting between 3dr services, droidplanner or andropilot. But before posting here, I actually Uninstalled Andropilot and Droidplanner, and even rebooted as it was a Microsoft device :)  the problem remains.

Can anyone say if "connection failed" is caused by missing mavlink heartbeat, or if it's somehow checking radio using AT commands first ?  I've also tried using standard sik.1.9 firmware

ne0fhyk, 2015-05-24:
`Connection failed` usually implies the app is not able to talk/access the connected usb device. If the mavlink heartbeat was missing, it would be able to connect, but would start broadcasting the `Data link lost...` message.

What (microsoft) device are you using to attempt the connection?

AndKe, 2015-05-24:
Ok, thank you - that points me in the (hopefully) right direction.  - so the problem is it does not see USB device ?  that would be on a enumeration PID/VID level ?  - or does it also check some AT commands ?

referance to mirosoft was just a bad joke due to desperate rebooting. these are Android (4.4+ devices)  - no microsoft here.

The device is aware of the OTG/USB device - (I see it when connecting the radio, a popup asks to use it for 3DR services.)   - so OTG and enumeration is ok.

AndKe, 2015-05-24:
BTW: upon connecting USB OTG+radio , then allowing 3DR services to use the device, no application pops up, (is that normal? )
I can start 3DR services, and from there launch tower, or just launch tower - but it does not change the outcome.

(I've also cleared application data for 3DR Services and Tower.)

ne0fhyk, 2015-05-24:
Yes, that's normal. Once the permission request pops up, can you try setting 3DR Services to `Always`.
Once that's set, attempt to connect Tower again. 

There's a bug that causes the usb connection to fail the first time after receiving the usb access.
Once the `connection failed` message shows, attempt to connect a second time through Tower.

AndKe, 2015-05-24:
- yep - I've set it as default application to handle this USB device on
both devices, same problem :(
BTW: the spelling of the error is "connection failed:"   - should there be
more after ":"  - or does that give a clue at all ?

ne0fhyk, 2015-05-26:
@AndKe Can you make sure the connection type is `USB`. The typical error for a `USB` connection type is `Connection failed: No Devices found`.

`Connection failed:` with no additional info usually occurs with the `UDP` connection type.

AndKe, 2015-05-26:
Tower defaults to USB, but now I've verified that even switching to TCP, then back, fails.

AndKe, 2015-05-28:
I sniffed the byte tower sends upon pressing "connect".
(radios was set at 57600 , as usual) but decoding it does not make any sense , instead of text - this is what Tower sends 
0A FE 09 FF BE 06 60 32 FE 09 01 FF BE 06 8A 4C FE 09 02 FF BE 06 B4 CF FE 09 03 FF BE 06 5E B1 FE 09 04 FF BE 06 D9 C1 FE 09 FF BE 06 33 BF

what exactly is "Tower" sending when pressing Connect ? - I'd assume it tries to AT commands - but after 0A (linefeed) we are way past ASCII.. 
Same string is 100% reproducable at any capture.

arsenixprime, 2015-06-12:
I have the exact same issue as AndKe. I had just posted on ardupilot forum http://ardupilot.com/forum/viewtopic.php?f=15&t=12724

I have also captured the same serial strings from a remote PC w/3DR radio when hitting the "connect" button in DP. This confirms that DP is sending data correctly through USB and radio.

The odd part is that if I connect a PC to the vehicle with MP, it works fine and I can see that the vehicle starts streaming telemetry down (by flashing light). If I then connect with DroidPlanner it works every time. The issue seems to be that whatever DP is sending to initiate the connection with the vehicle is getting rejected/ignored.

arsenixprime, 2015-06-12:
I also forgot to mention... once I try to connect with DroidPlanner I have to reboot the vehicle to get MP to work.

Here's my theory: I think DroidPlanner is putting the remote radio into command mode or some other mode where traffic is blocked from the PixHawk. I have seen it happen with MP occasionally too (a reboot clears it), but it happens every time with DP. Alternatively maybe DP is causing the autopilot to stop sending heartbeats?

I am not sure why DP is sending that traffic after hitting connect. MP's behaviour when you connect is to wait until a MAVLINK heartbeat packet comes through and then respond, it sends nothing until it hears from the vehicle.

AndKe, 2015-06-12:
uploaded Android log, hopefully useful to good java developers :)
some comments/markers in all caps plus extra linefeeds.

https://gist.github.com/AndKe/060209e7aec7f9a3d6bd
in short, USB init (on clicked connect) , and right then all goes south in the same moment: 
```
06-12 17:00:14.755 D/UsbRequestJNI(29278): init
06-12 17:00:14.755 I/System.out(29278): First heartbeat timeout.
06-12 17:00:14.785 D/DroidPlannerService(29278): Drone manager disconnection for org.droidplanner.android
06-12 17:00:14.785 D/DroneManager(29278): Disconnecting client org.droidplanner.android
06-12 17:00:14.795 D/DroidPlannerService(29278): Disconnecting...
06-12 17:00:14.795 D/ProcessRequestThread::(29278): Device has been closed.
06-12 17:00:14.795 W/System.err(29278): java.lang.InterruptedException
06-12 17:00:14.795 W/System.err(29278): 	at java.util.concurrent.locks.AbstractQueuedSynchronizer.doAcquireSharedInterruptibly(AbstractQueuedSynchronizer.java:975)
06-12 17:00:14.795 W/System.err(29278): 	at java.util.concurrent.locks.AbstractQueuedSynchronizer.acquireSharedInterruptibly(AbstractQueuedSynchronizer.java:1281)
06-12 17:00:14.795 W/System.err(29278): 	at java.util.concurrent.Semaphore.acquire(Semaphore.java:283)
06-12 17:00:14.795 W/System.err(29278): 	at com.ftdi.j2xx.ProcessInCtrl.acquireReadableBuffer(FT_Device.java:2028)
06-12 17:00:14.795 W/System.err(29278): 	at com.ftdi.j2xx.ProcessRequestWorker.run(FT_Device.java:2496)
06-12 17:00:14.795 W/System.err(29278): 	at java.lang.Thread.run(Thread.java:841)
```

arsenixprime, 2015-06-12:
I think that log looks "normal" for timing out on the heartbeat.

Is the serial capture above the complete data that was sent? Those packets are Mavlink but they are incomplete. The payload length is the second byte ("09"), but the messages should then be 17 bytes total. When I sniffed mine they were similar but were the correct length.

AndKe, 2015-06-12:
yes, the data above is what I get every time, no wonder if it's a bit short, the log I posted shows only 30ms between init and the fail.

arsenixprime, 2015-06-12:
Oh... hmm. That seems like maybe a different issue than what I am experiencing. My Droidplanner waits about 5 seconds before it says fail. During that time it sends out about 6 Mavlink packets which are similar to what you posted, one every second or so. They are complete packets though not fragmented like yours.

AndKe, 2015-06-12:
yes, the time it takes here is the same here too, but I get only those bytes, all at once, I just split them into lines for readability.  
It's the log that proves to me that it happens at once,  - please check your logs - install some logcat /catlog application for Android - then you'll see - I bet you have same issue  :)

arsenixprime, 2015-06-13:
It seems my issues are resolved (for now). I am a bit suspicious that they will reappear but I have not been able to get the fault in 24 hours. I changed the channel on all my radios (from the default), under the theory that maybe the reason I was getting randomly inconsistent results might be because someone else is playing around with 3dr radios in the area on the default channel. I also disabled flow control on my vehicles, on the theory that noise on the flow control lines might be causing it to be erroneously auto detected (which would cause the symptom I was seeing of the autopilot failing to send data until rebooted). None of these are really DroidPlanner specific, so I am still unsure why it consistently failed with DP and not on the PC.

AndKe, 2015-07-04:
The failure is caused by using HW flow control - see final comments here: http://ardupilot.com/forum/viewtopic.php?f=15&t=12724&start=10

brad112358, 2015-08-21:
If this issue was indeed caused by the bug in ardupilot's automatic flow control, it should be fixed in AC3.3rc9.  Perhaps someone who experienced this problem would like to test with that release and report back here so this can be closed as well.

AndKe, 2015-08-21:
Installed AC3-3rc9 by enabling beta builds , I did *not* build from master.
So unless the fix is only in master, not in rc9  - I am sorry to say the issue remains, I see no difference in Tower behaviour.

AndKe, 2015-08-28:
Update:  (now tested with AC3.3rc9)
If the Android-side radio has HW flow control enabled, there is no longer an error message, just no communication.

cuchara, 2016-08-24:
hello !!!  here I have the same problem

You find solution ??