DroidPlanner/Tower #1488: Mission Upload to APM not working in Tower. No prob in DroidPlanner 2, though.

RipperDrone, 2015-05-25:
Wierd: Mission Upload is not an issue in DP 2. However, in Tower I can only download missions, not upload any. Just no response, nothing. 

I'm using 3DR telemetry v2.

Mission Planner on PC is transmitting missions / waypoints flawlessly as well.

What's wrong here, anybody made a similar observation?
All other communications with Tower seems to be fine.

Pl help me out.
RipperDrone

RipperDrone, 2015-05-25:
Tried a couple of things. Finally, deinstalling / reinstalling Tower did the magic! Not sure what was going on there. Now both apps can upload missions to the UAV. ;-))

closed 2015-05-25 by ne0fhyk

JaysonBlack, 2016-08-25:
ah ok. i will try that too. i am not able to upload a mission. nothing happens after selecting upload. perhaps developer could build something into the code that warns if mission did not successfully load into drone?!?!? and even better give the user a clue as to what may have gone wrong :D