DroidPlanner/Tower #1490: Feature request: hybridised Auto/Mission + Follow Me/Lead It - a virtual tether on a predefined course

Felixrising, 2015-05-28:
Follow Me Mode is a great addition to the APM:Copter and Tower (DroidPlanner 3) specifically, however I'd like to suggest a hybridized Auto-Mission + Follow Me/Lead It mode to increase it's usefulness.

How would this work?
A set course can be laid out including elevations via typical Mission planning, however the App would be virtually "tethered" to the APM device effectively controlling speed with vector defined by the pre-planned mission. This way the Tower app simply pushes or pulls the APM device around the course... 

Building on this new feature would be POI, or other pre-programmed or triggered actions based on location or even speed thresholds or tether "push/pull" as trigger events.

Thalek, 2015-05-28:
I proposed something roughly similar once, which I called "Lead Me."  My thought was that if you spotted something interesting on a reconnaissance flight (recorded or FPV), you could then have the UAV lead you to the POI, probably orbiting the location once it arrived.

floydmueller, 2015-06-01:
Great idea!