DroidPlanner/Tower #1492: Is connecting directly to the Drone via USB supported?

Teemperor, 2015-05-30:
I can't find any documentation on that topic (only connecting to the phone side of the telemetry via cable), but I want to connect to the Pixhawk via a micro usb cable from my phone. Is that possible?

Felixrising, 2015-05-31:
Yup. You just need a USB OTG cable. It's not documented in the <a href=https://github.com/DroidPlanner/Tower/wiki/How-to-connect>wiki</a> but it was added back int v0.12.

hamishwillee, 2015-05-31:
@Teemperor - if that works for you, update the wiki?

Teemperor, 2015-06-03:
@hamishwillee it doesn't work for me. I get this error back from the tower app https://docs.google.com/file/d/0B3kxMHEUP2-lb1hWX2xBUHYtWjA/edit?usp=docslist_api

Teemperor, 2015-06-03:
I tried with a usb OTG cable and a normal usb-to-micro-usb cable. Phone was a Moto G. I'll try to get my hands on a tablet with a non-stock ROM to see if it works there.

Teemperor, 2015-06-03:
Ok, my bad, the usb-to-micro-usb was faulty and it worked with a different one.

Teemperor, 2015-06-03:
Added documentation to the wiki.

closed 2015-06-03 by Teemperor

hamishwillee, 2015-06-03:
@Teemperor - so can you write up a brief addition to the wiki? Of course I could add this as a case that works, but you can show your actual setup.

Teemperor, 2015-06-04:
@hamishwillee So, should I add the used hardware specs to [my section](https://github.com/DroidPlanner/Tower/wiki/How-to-connect#usb-to-the-apmpixhawk-without-telemetry) in the "How to connect" article or somewhere?