DroidPlanner/Tower #1494: Force disarm of copter by sending DISARM command with param2 with 21196

rmackay9, 2015-06-03:
AC3.2.1 and AC3.3 require the vehicle to be landed before the vehicle will accept a disarm request.
AC3.3 will accept the disarm however (regardless of whether the vehicle is landed or not) if param 2 is set to 21196

hamishwillee, 2015-06-03:
... as documented in [MAV_CMD_COMPONENT_ARM_DISARM](http://copter.ardupilot.com/wiki/common-planning-a-mission-with-waypoints-and-events/common-mavlink-mission-command-messages-mav_cmd/#copter-2)

squilter, 2015-06-26:
With the addition of DroidPlanner/DroneKit-Android#172, this is easy to implement.  We just need to find a good place to put the button.

ne0fhyk, 2015-06-26:
This is already implemented:
https://github.com/DroidPlanner/Tower/issues/1433

The implementation needs to be updated to make use of the new capability.

squilter, 2015-07-07:
I will implement this shortly.  See issue #1433

closed 2015-07-07 by squilter