DroidPlanner/Tower #1495: Are autopilot warnings broken?

chapman, 2015-06-03:
Hi, I have a copter which frequently fails the pre-arm checks, previously the app would announce what the problem was, but now doesn't. 

I see the option under preferences to enable/disable warning messages from the autopilot, it is definitely checked (and I tested with it off in case the sense was reversed) but can't ever get Tower to announce these critical messages from the autopilot using the current build (v3.1.6).

Running on a Nexus7 with Android v5.0.2

Can anyone else confirm this? (simple repro: shake your copter during the red/blue flashing light sequence while first powering up, then try arm)

chapman, 2015-06-03:
Small update/hint: it does report "Safety Switch" error when arming, but not "gyro health" or "alt disparity"

squilter, 2015-07-06:
@chapman The error reporting has changed slightly in the flight code since 3.1.6.  Could you see if this problem still exists with AC3.2 or even 3.3?

squilter, 2015-07-08:
Nevermind, I see it.  Something is different about AC3.3.  Tower only reports `rc failsafe` and `need gps lock` with 3.3.  AC3.2 and 3.1 report all errors properly.  I will investigate.

ne0fhyk, 2015-07-29:
Fixed by pr #1550

closed 2015-07-29 by ne0fhyk