DroidPlanner/Tower #1497: "400ft exceeded" warnings with Plane (altitude AMSL)

github-0, 2015-06-09:
Hi,

Tried Tower for the first time today with Pixhawk/Plane 3.3.0. Tower seems to display the Above Mean Sea Level altitude instead of home altitude and in my case this triggers a loop of 400ft exceeded warnings. I did disable the warning, but it is higly inconvenient that the HUD shows AMSL instead of AGL. Any chance to get this changed or show both?

ne0fhyk, 2015-06-09:
@github-0 thanks for the feedback. ArduPlane behaves differently from ArduCopter by reporting the AMSL instead of AGL, and Tower just forward that information along.

github-0, 2015-06-10:
From what I can understand, this same issue was solved in Mission Planner here https://github.com/diydrones/MissionPlanner/issues/40 (also https://github.com/diydrones/ardupilot/issues/417). They changed the HUD field to GLOBAL_POSITION_INT.relative_alt field so that the behavior is consistent between copter and plane.

ne0fhyk, 2015-08-19:
@github-0 thanks for the info. The issue is being fixed through #1517, and the next beta release will incorporate the fix.

closed 2015-08-19 by ne0fhyk