DroidPlanner/Tower #1506: GPS not available

oseiler2, 2015-06-29:
The app isn't acquiring a GPS location, despite GPS being turned on, locked, and working with other apps. 
This happens on my Nexus 7 (2012) on Android 5.1.1 as well as on my SGS 4 on Android 5.0.1.

For the S4 I've found a work-around: when using another app (e.g. Google's My Tracks) that uses the GPS in the background then Tower also receives location updates. Unfortunately this doesn't work on the Nexus 7.

vvv, 2015-08-30:
This issue is reproducible on Nexus 7 2013, Android 5.1.1.
Tower-v3.1.6; 3DR Services v1.3.3.

ne0fhyk, 2015-08-31:
@oseiler2 @vvv  what are the symptoms for this bug (i.e: no user location update on the map, unable to start follow me...)

oseiler-mbie, 2015-08-31:
Symptoms: the user location on the map is in the wrong place. The app seems to think is has a GPS location, but it does not. The GPS icon that you can normally see in the status bar when an app consumes GPS is not present (unless I use the workaround above on my S4 - using another app in background that causes the GPS to turn on).

ne0fhyk, 2015-08-31:
@oseiler-mbie the app is behaving as designed, and that's not a bug. While on the flight data screen, the app selects the [balanced power accuracy gps setting](https://developers.google.com/android/reference/com/google/android/gms/location/LocationRequest.html#PRIORITY_BALANCED_POWER_ACCURACY) in order to reduce battery consumption.

oseiler-mbie, 2015-08-31:
I understand, but should it not change to acquire an accurate GPS position as soon as I start flying? Or using follow me? Both don't work for me :(

ne0fhyk, 2015-08-31:
It doesn't change once the vehicle is active, but does when follow me is activated.
Does the follow me icon remains 'orange' when enabled?

oseiler2, 2015-08-31:
Sorry, I don't remember, Last time I put it in follow me the app thought I was elsewhere and my quad started moving awaw from me until I deactivated follow me. Happy to check next time I fly. What is the expected outcome? What trigger the more accurate GPS settings (when should I see an accurate position and when a coarse position)?

oseiler2, 2015-08-31:
And... keep in mind that on my Nexus 7 I cannot get a good position, even when I habe another app using the GPS in accurate mode in the background.

vvv, 2015-08-31:
The problem is not limited to Tower. Google Maps and Earth don't update location as well. GPS positioning is not operational.
See [GPS signal lost after update 5.1.1](https://productforums.google.com/forum/#!topic/nexus/aZIuXQokLbw). (‘GPS Status and Toolbox’, recommended by one of the comments there, cannot find location either.)

ne0fhyk, 2015-08-31:
For now, only follow me triggers a high accuracy gps position request.
If it's a desired feature, the default behavior can be updated to request a high accuracy as well, but bear in mind that might impact the device battery life while using the app.

vvv, 2015-08-31:
@ne0fhyk: This looks like Android's bug, not Tower's.
See the [bug in action](http://youtu.be/bYM--XtCfgI).

jeepy, 2015-09-09:
Hi, 
When I use an old version of Droidplanner the phones gps works perfectly and very accurate every time. 
Only when I use Tower the problem that the gps is way off acures as described above. 
I'm using Sony Z3 compact with Android 4.4.4 

Do I undetstand this ccorrecty, only when I enable follow me then tower is switching the phones gps into high accuracy mode ?

ne0fhyk, 2015-09-14:
I'm updating the issue's title to reflect the actual problem.

ne0fhyk, 2015-09-14:
High accuracy gps mode has been enabled by commit f8863bac525ab06d454bda15b7527cc3f0c6068d

closed 2015-09-14 by ne0fhyk