DroidPlanner/Tower #1509: Mobius Action Camera Lens C

sbcrumb, 2015-07-06:
Can anyone help me with the Specs for this Camera? And can we get it added to Tower?

<cameraInfo>
  <Name>Mobius Lens C</Name>
  <SensorWidth>??</SensorWidth>
  <SensorHeight>??</SensorHeight>
  <SensorResolution>??</SensorResolution>
  <FocalLength>??</FocalLength>
</cameraInfo>

FLYBYME, 2015-07-14:
```
<?xml version="1.0" encoding="utf-8"?>
<cameraInfo>
    <Name>Mobius A lens</Name>
    <SensorWidth>5.07</SensorWidth>
    <SensorHeight>3.38</SensorHeight>
    <SensorResolution>5</SensorResolution>
    <FocalLength>??</FocalLength>
</cameraInfo>
```

I dont know what the focal length would be..

sbcrumb, 2015-07-14:
Is this the specs for the C version lend?

FLYBYME, 2015-07-14:
I believe the sensor is the same.

ne0fhyk, 2015-09-14:
@sbcrumb @FLYBYME are these specs complete?

closed 2015-09-26 by ne0fhyk