DroidPlanner/Tower #1510: Param auto-upload?

squilter, 2015-07-08:
The current process for uploading parameters is to change all desired parameters in the list, then to click "upload" in the overflow menu.

Some users have reported that they didn't realize they needed to click the upload button.  They would change the parameter in the list, and then go back to the flight page without having uploaded the parameters.  

It got me thinking:  is there any reason why Tower doesn't just upload the parameters automatically?

ne0fhyk, 2015-07-08:
Mostly ux, any action that makes permanent changes needs to be confirmed by
the user (I.e: delete action, etc...).
At best we can improve the ux to promote the upload action after they
modify a parameter.

TShapinsky, 2015-07-23:
fixed by #1535

closed 2015-07-23 by squilter