DroidPlanner/Tower #1517: Investigate plane support

ne0fhyk, 2015-07-13:
There are reports of missing/buggy plane features.

hamishwillee, 2015-07-13:
There are missing features, but I suspect that most of them are due to limitations in Plane's support for guided mode. According to Tridge you'd be more likely to write a script that uses AUTO mode and dynamically update the mission on the fly.

Takeoff is the main limitation. The problem is more than just missing pitch angle - Plane does not support takeoff in guided mode full stop. The recommended way to take off is to start an AUTO mode mission with a takeoff command in it. After taking off you can switch to guided mode if you want.

Upshot, I suspect we're going to have to request features in plane to get any sort of consistent experience.

ne0fhyk, 2015-08-05:
@squilter I'm leaving this issue assigned to you. Play around with Tower and `SITL` or a plane, and report what you find. 
As @hamishwillee mentioned, some of the issues might need to be addressed at the flight code level.

timmmt, 2015-08-10:
I've just finalized my first fixed wing setup with Pixhawk, noticed that Tower does not display all the plane flight modes, should I file a new bug (did not find any atm) or which is the recommended way to keep the feedback coming?

squilter, 2015-08-10:
@timmmt Can you just note all plane problems in this issue?  I think there will probably be quite a few.

timmmt, 2015-08-10:
Thanks, I do have to take back my previous comment, figured out when powering on the plane it was in RTL = focus set lower part of the list. Scrolling down did not work and user was confused. Still user error.

squilter, 2015-08-13:
I am gettting a constant "warning: 400 feet exceeded".  Over and over again.  Even though I'm at 2000ft.  Shouldn't I only get that when I cross to 400 ft?

squilter, 2015-08-13:
Auto missions are hard.  For some reason a waypoint was automatically created at -656 feet.  I tried to increase the altitude to 2000 ft. (ground was aroun 1900ft).  Took a very long time to keep swiping the altitude slider.  And then the slider wouldn't let me go above 656 ft.

squilter, 2015-08-13:
Mavproxy is reporting altitude above ground level, so it is clearly possible.  We should try to do that, and just ignore altitude above sea level.  That will solve lots of problems.

squilter, 2015-08-13:
Pause button doesn't work if I click it from loiter mode.  But if I switch to guided, then the pauuse button selects itself.  I'm not sure if pause is too useful with plane.  It's exactly the same thing as loiter mode as far as I can tell.

ne0fhyk, 2015-08-13:
- [ ] Disable killswitch on plane

justicebentz, 2015-08-13:
I haven't tested new Beta Tower on arduplane yet, will do that this weekend and provide feedback.

Load multiple missions in one flight. 
Enable killswitch for plane 

IE: Once plane returns from first auto mission, load a new mission and fly that... Currently I have to land plane and reboot to do second mission. 

Again thank you for your contributions to the app, keep us arduplane users in mind :)

justicebentz, 2015-08-14:
I am not receiving RSSI signal on my GCS?? 

[image]

squilter, 2015-08-14:
@justicebentz Are you using standard 3dr telemetry radios, or something else?  Plane or Copter?

justicebentz, 2015-08-14:
RFD 900 with boosted antenna's 
[image]

squilter, 2015-08-14:
I *think* the rfd9000 will show that data.  I think the bluetooth bridge is what's messing that up.  Can you just connect directly to the rfd9000 from the phone and see what happens

justicebentz, 2015-08-14:
It had worked prior to the beta with rfd900 and Bluetooth was working. Had
70 percent rssi at 7 mile range

squilter, 2015-08-18:
Takeoff and landing aren't supported in guided.  It is slightly more complicated because takeoff requires a final altitude AND an angle.  Landing is also more complicated because it requires a landing point.  It can't realistically just land at whatever coordinate it's flying at right now.

For altitude we should be reading `GLOBAL_POSITION_INT.relative_alt*0.001`, not VFR HUD.  In fact, Tridge thinks that Copter isn't following the protocol, and that we should always be listening to `GLOBAL_POSITION_INT.relative_alt*0.001` for relative alt.

ne0fhyk, 2015-08-19:
@justicebentz found the issue with the lack of signal info. It'll be fixed with the next 3DR Services beta release.

closed 2015-09-14 by ne0fhyk